shmuelzon/esp32-ble2mqtt #182: Support repeating UUIDs

dakhnod, 2023-02-21:
I am building a peripheral supporting the Automation IO service, which could lead to multiple characteristics within one service with identical UUIDs, which is perfectly fine with the BLE specification.
That does seem to break your code, though.
My suggestion would be to publish repeating UUID's with a suffux, e.g. "0x0001", "0x0001_1", "0x0001_2", etc.

Thanks for your work in any case,
feel free to discuss the idea with me.

shmuelzon, 2023-02-22:
I knew a requirement like this would come by sooner or later... :)

I'm generally OK with adding suffixes to repeating UUID to distinguish between the different ones but it would be quite a code change for all the functions that rely on the characteristic's UUID only. Now they'll need something else in addition like the index or whatever of the repeating characteristic.
I understand that repeating characteristics will have different handle numbers but do you know if they have a single "characteristic client config" or one per repeating characteristic?

dakhnod, 2023-02-23:
I can confirm that every characteristic has it's own subset of descriptors. They are not shared, even if the UUID repeats.

shmuelzon, 2023-02-23:
OK, I'll see what I can do on a side branch and will probably need your help to test/debug as I don't have any devices with repeating UUIDs

dakhnod, 2023-02-23:
If you want you can leave the implementation to me, since I already have experience with that. Also, I am somewhat familiar with your code already.

shmuelzon, 2023-02-23:
That would probably get it done faster :)
I'm, basically, imagining adding an index field to each characteristic and the functions that handle characteristics (converting it to a name and back, etc.).
If you have any other thoughts or would like to share your work even before it's complete, I'd be happy to take a look.

dakhnod, 2023-02-25:
Got it: https://github.com/shmuelzon/esp32-ble2mqtt/pull/183

shmuelzon, 2023-03-31:
Closed by #183

closed 2023-03-31 by shmuelzon