shmuelzon/esp32-ble2mqtt #187: ESP32s3

JuGlass, 2023-03-09:
Hello ! 

I recently upgraded my esp32 to an esp32s3 but after several tries with Espressif ide 5.0 and visual code with your recommendations for versions, I realized that I can't build an image for esp32s3 but for esp32 it works fine.

in my logs I find this error " warning : 'eth_netif' defined but not used [-Wunused-variable]
   19 | static esp_netif_t *eth_netif = NULL;"

Do you know how I can solve this problem? 

Thank you for your project 
Regards

cyberhuman, 2023-03-09:
The fix would be to add
```
#if CONFIG_ETH_USE_ESP32_EMAC
...
#endif
```

around the line 19 in eth.c.

JuGlass, 2023-03-10:
@cyberhuman thank you for your answer unfortunately it does not solve the problem I still can't build an image

shmuelzon, 2023-03-31:
Hey, I fixed the build but I don't have an ESP32-S3 so I can't verify it actually works...
Please test and let me know

closed 2023-03-31 by shmuelzon

JuGlass, 2023-07-07:
Hey, 
i didn't saw the updates in my email 

To give you a feedback
i can not have any data from the ble device 
it"s says connected i can also see on the device it"s connected hbut then nothing no data transfert and i didn't change any setting from the device 

see in the log below 

I (3695) MQTT: Connecting MQTT client
I (4665) MQTT: MQTT client connected
I (4665) BLE2MQTT: Connected to MQTT, scanning for BLE devices
I (4855) BLE2MQTT: Discovered BLE device: 57:cd:0e:f2:fc:7a (RSSI: -85), connecting
I (4855) BLE2MQTT: Discovered BLE device: e3:5b:33:e8:f2:fa (RSSI: -64), connecting
I (5175) BLE2MQTT: Discovered BLE device: 5d:75:6c:d2:2b:13 (RSSI: -79), connecting
I (5565) BLE2MQTT: Discovered BLE device: 2b:5e:a8:aa:34:a8 (RSSI: -67), connecting
I (5575) BLE2MQTT: Discovered BLE device: 7b:33:47:51:1b:26 (RSSI: -65), connecting
I (5885) BLE2MQTT: Discovered BLE device: d6:74:3c:b6:4b:c3 (RSSI: -63), connecting
I (6365) BLE2MQTT: Discovered BLE device: 26:8e:da:a0:aa:6a (RSSI: -51), connecting
I (6435) BLE2MQTT: Discovered BLE device: 63:f5:d7:96:29:04 (RSSI: -64), connecting
I (6445) BLE2MQTT: Discovered BLE device: 77:5c:c7:0e:e1:2f (RSSI: -81), connecting
I (6685) BLE2MQTT: Discovered BLE device: 18:db:ed:55:6c:67 (RSSI: -47), connecting
I (6915) BLE2MQTT: Discovered BLE device: f7:eb:ed:88:d4:b4 (RSSI: -81), connecting
I (6995) BLE2MQTT: Discovered BLE device: 1b:26:9f:21:d0:69 (RSSI: -76), connecting
I (7395) BLE2MQTT: Discovered iBeacon broadcaster
I (7405) BLE2MQTT: Discovered BLE device: 28:de:65:a9:dc:4e (RSSI: -88), connecting
I (7405) BLE2MQTT: Discovered BLE device: 71:85:30:00:93:c8 (RSSI: -69), connecting
W (37915) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x100
E (37915) BLE: Open failed, status = 0x85
I (42415) BLE2MQTT: Connected to device: e3:5b:33:e8:f2:fa, scanning
I (46375) BLE2MQTT: Connected to device: 5d:75:6c:d2:2b:13, scanning
W (71025) BT_APPL: gattc_conn_cb: if=1 st=0 id=513 rsn=0x100
E (71025) BLE: Open failed, status = 0x85
W (71775) BT_APPL: gattc_conn_cb: if=1 st=0 id=257 rsn=0x13
W (71775) BT_HCI: hcif disc complete: hdl 0x2, rsn 0x13
I (71785) BLE: Connection closed, reason = 0x13
I (71785) BLE2MQTT: Disconnected from device: 5d:75:6c:d2:2b:13
I (76745) BLE2MQTT: Connected to device: 7b:33:47:51:1b:26, scanning

 57:cd:0e:f2:fc:7a it the only device withelisted in the config JSON but still  try to connect to other devices

regards
Ju

shmuelzon, 2023-07-09:
@JuGlass, can you please post your configuration file (while removing sensitive information)?

commit 2753ca7f (2023-03-31): Fix build for ESP32-S3 and ESP32-C3

Fixes #187
PR #189 (2023-03-31): Fix build for ESP32-S3 and ESP32-C3
Fixes #187
commit da8e2c22 (2023-03-31): Fix build for ESP32-S3 and ESP32-C3 (#189)

Fixes #187