shmuelzon/esp32-ble2mqtt #192: Configure Scanning?

giddyhup, 2023-04-07:
**Question**
I am not sure if this is a "works as designed", a bug situation, or if I am imagining it, therefore I decided to enter it under General Questions. 

I have one device (a treadmill) whitelisted and that works well, yet it seems that the solution still interferes with another device's function. Notably, my Bluetooth earbuds disconnect while the audio of the device they are supposed to play back from (an Echo device) is being paused. To put it differently, while audio is paused they seem to react to queries from the ESP and disconnect from the proper device. Once playback continues they reconnect again.

Are my observations correct or is the solution neither scanning nor reacting to queries from other devices?

shmuelzon, 2023-04-20:
Hey, I can't argue with your observations but it does sound like odd behavior to me. The ESP is always in scanning mode which means it listens to advertisements (passive) but also sends out scanning requests (active). Even if it does find your earbuds, assuming those aren't whitelisted, it shouldn't try to connect to them.
Have you tried pausing audio and using a mobile phone to scan for Bluetooth devices to see if it behaves any differently?
We may be able to tune some of the scanning parameters but. to be honest, I have no idea what exactly.

giddyhup, 2023-05-07:
Sorry for letting this dangle for so long but I have more observations and one possible approach to either fix it or at least take esp32-bl2mqtt out of the equation.  

When another household-member is using the treadmill a different kind of earbuds is being used. Here, there are no disconnects but the treadmill is acting up. Acting up in this case means that sometimes it switches from running mode to pause mode (slowing down). Both issues, the earbud-disconnect and the treadmill-pausing do not happen when esp32-bl2mqtt is not being used.

The earbud-disconnect I can live with since it does not happen once audio is streaming. The treadmill-pausing is more annoying (and less explainable).

Would it be possible to add an option where only the whitelisted device is communicated with, other advertisements are ignored and no scanning requests are being sent? Some kind of stealth mode?

shmuelzon, 2023-05-09:
> Would it be possible to add an option where only the whitelisted device is communicated with, other advertisements are ignored and no scanning requests are being sent?

The ESP needs to actively scan to detect the devices it may want to connect with. This scanning request, AFAIK, is a generic broadcast and not destined to any specific BLE device. It then starts listening to any device that might respond and only if that device is in the whitelist, does it try to connect to it.
Advertisements are passive by nature, the ESP doesn't request them and if it acts upon receiving them, it's only to send out something on the MQTT bus, nothing on the BLE side.

If you're able to test something out, I can suggest changing the scanning mode from active to passive.
This can be done by editing [main/ble.c:22](https://github.com/shmuelzon/esp32-ble2mqtt/blob/da8e2c2230e38439a15a85275bba97f4da774521/main/ble.c#LL22C18-L22C38) and changing `BLE_SCAN_TYPE_ACTIVE` to `BLE_SCAN_TYPE_PASSIVE`. I believe that in this mode it might not detect all BLE devices but it should also only send out BLE packets only to the devices in the whitelist.

Let me know...