shmuelzon/esp32-ble2mqtt #193: BLE Advertisment publishing?

ArcadeMachinist, 2023-08-01:
Hello!

I was recently playing with the project and can't find how to publish BLE advertisment packet, if it is possible at all.

I have a range of devices (RGB lights) that are controlled by publishing control packet as a manufacturer data of a BLE advertisment packet.

Is there any way to publish an advertisment?
Something like that can be achieved in Linux by "btmgmr adv-add -d <hexdata> 1"

shmuelzon, 2023-08-01:
Hey,

It's, currently, not possible to publish advertisement packets. Not sure how to even define the MQTT payload to do that...

ArcadeMachinist, 2023-08-08:
I would say just a JSON structure with either BLEAdvertisment1 fields for constructing the advertisment or another command just to register the raw advertisment data (sent in hex as a field in JSON).

mhaberler, 2023-08-08:
I think that'd be super useful - in particular when combined with say https://github.com/theengs/decoder server-side

that adds 70+ meaningful sensor reports out of the box: https://decoder.theengs.io/devices/devices.html