shmuelzon/esp32-ble2mqtt #198: BLE Scan / broadcasters crash device

dakhnod, 2024-04-17:
My BLE bridges on average every day. Sometimes more, sometimes less. After recording the log, it's always this sequence. Two devices in my surrounding seem to cause this:

```
[2024-04-16 16:48:05] I (30967920) BLE2MQTT: Discovered iBeacon broadcaster
[2024-04-16 16:48:13] I (30976160) BLE2MQTT: Discovered BLE device: 78:27:4e:ec:ef:bc (RSSI: -92), not connecting
[2024-04-16 16:48:16] I (30978480) BLE2MQTT: Discovered BeeWi Smart Door broadcaster
[2024-04-16 16:48:16] E (30978480) BT_LOG: Invalid p_eir data.
[2024-04-16 16:48:16] 
[2024-04-16 16:48:16] Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[2024-04-16 16:48:16] 
[2024-04-16 16:48:16] Core  1 register dump:
[2024-04-16 16:48:16] PC      : 0x400dc779  PS      : 0x00060d30  A0      : 0x800d854c  A1      : 0x3ffeb240  
[2024-04-16 16:48:16] A2      : 0x00000000  A3      : 0x00000000  A4      : 0xffffffa0  A5      : 0x400d7e88  
[2024-04-16 16:48:16] A6      : 0x3fff579c  A7      : 0x3ffc055c  A8      : 0x800dc779  A9      : 0x3ffeb220  
[2024-04-16 16:48:16] A10     : 0x00000000  A11     : 0x000000ff  A12     : 0x3ffeb240  A13     : 0x00000000  
[2024-04-16 16:48:16] A14     : 0x00000001  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
[2024-04-16 16:48:16] EXCVADDR: 0x00000003  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  
[2024-04-16 16:48:16] 
[2024-04-16 16:48:16] 
[2024-04-16 16:48:16] Backtrace: 0x400dc776:0x3ffeb240 0x400d8549:0x3ffeb270 0x400d870d:0x3ffeb3d0
[2024-04-16 16:48:16] 
[2024-04-16 16:48:16] 
[2024-04-16 16:48:16] 
[2024-04-16 16:48:16] 
[2024-04-16 16:48:16] ELF file SHA256: 9617debcb72431e5
[2024-04-16 16:48:16] 
[2024-04-16 16:48:16] Rebooting...
```
and
```
[2024-04-16 19:46:39] I (10702910) BLE2MQTT: Discovered BLE device: c5:69:d1:9f:6d:e0 (RSSI: -100), not connecting
[2024-04-16 19:46:39] I (10702990) BLE2MQTT: Discovered BLE device: 4b:bd:43:d5:9d:35 (RSSI: -94), not connecting
[2024-04-16 19:46:47] I (10710980) BLE2MQTT: Discovered BLE device: 82:77:16:e6:78:d0 (RSSI: -97), not connecting
[2024-04-16 19:46:47] I (10710980) BLE2MQTT: Discovered Xiaomi Mijia broadcaster
[2024-04-16 19:46:47] E (10710990) BT_LOG: Invalid p_eir data.
[2024-04-16 19:46:47] 
[2024-04-16 19:46:47] Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[2024-04-16 19:46:47] 
[2024-04-16 19:46:47] Core  1 register dump:
[2024-04-16 19:46:47] PC      : 0x400dcea9  PS      : 0x00060330  A0      : 0x800d854c  A1      : 0x3ffeb0f0  
[2024-04-16 19:46:47] A2      : 0x00000000  A3      : 0x00000000  A4      : 0xffffff9a  A5      : 0x400d7e88  
[2024-04-16 19:46:47] A6      : 0x3fff3ff0  A7      : 0x3ffc0584  A8      : 0x800dcea1  A9      : 0x3ffeb0d0  
[2024-04-16 19:46:47] A10     : 0x00000000  A11     : 0x00000016  A12     : 0x3ffeb100  A13     : 0x00000000  
[2024-04-16 19:46:47] A14     : 0x00000001  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
[2024-04-16 19:46:47] EXCVADDR: 0x00000003  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  
[2024-04-16 19:46:47] 
[2024-04-16 19:46:47] 
[2024-04-16 19:46:47] Backtrace: 0x400dcea6:0x3ffeb0f0 0x400d8549:0x3ffeb280 0x400d870d:0x3ffeb3e0
[2024-04-16 19:46:47] 
[2024-04-16 19:46:47] 
[2024-04-16 19:46:47] 
[2024-04-16 19:46:47] 
[2024-04-16 19:46:47] ELF file SHA256: 9617debcb72431e5
[2024-04-16 19:46:47] 
[2024-04-16 19:46:47] Rebooting...
```

Is there any way to disable BLE broadcasters?

shmuelzon, 2024-04-20:
Hey,

There isn't a configuration option for this but, if you're up to building this project yourself, you could make the following changes:
```patch
diff --git a/main/broadcasters.c b/main/broadcasters.c
index 3ff9ce9..417a59f 100644
--- a/main/broadcasters.c
+++ b/main/broadcasters.c
@@ -765,11 +765,6 @@ static broadcaster_ops_t atc1441_temp_hum_ops = {

 /* Common */
 static broadcaster_ops_t *broadcaster_ops[] = {
-    &ibeacon_ops,
-    &eddystone_ops,
-    &mijia_sensor_ops,
-    &beewi_smart_door_ops,
-    &atc1441_temp_hum_ops,
     NULL
 };

```

dakhnod, 2024-04-20:
Should I open a PR with an option to disable broadcasters?

shmuelzon, 2024-04-21:
To be honest, I'd much rather fix the bug. I'd need to see the actual payload of the broadcast packet, though, to understand what's going on.

dakhnod, 2024-04-21:
I agree. It only happens sometime, sometimes the same device goes through ...

Matthias-Hess, 2025-08-03:
I ran into the exact same issue, my device keeps crashing and is knocked out because of a BeeWi Smart Door broadcaster. This bug deserves to be fixed.

I (3672) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3682) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (3692) WiFi: Connected
I (3712) wifi:<ba-add>idx:0 (ifx:0, ce:ac:94:ac:86:9f), tid:0, ssn:3, winSize:64
I (3742) wifi:<ba-add>idx:1 (ifx:0, ce:ac:94:ac:86:9f), tid:7, ssn:2, winSize:64
I (3752) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4692) esp_netif_handlers: sta ip: 192.168.8.203, mask: 255.255.255.0, gw: 192.168.8.1
I (4692) BLE2MQTT: Connected to the network, connecting to MQTT
I (4692) MQTT: Connecting MQTT client
I (4712) MQTT: MQTT client connected
I (4712) BLE2MQTT: Connected to MQTT, scanning for BLE devices
I (4832) BLE2MQTT: Discovered BLE device: df:de:b4:eb:ca:f4 (RSSI: -56), not connecting
I (4852) BLE2MQTT: Discovered BLE device: c5:91:9f:a8:b3:8f (RSSI: -46), not connecting
I (4862) BLE2MQTT: Discovered BLE device: ea:4d:d2:5c:27:cb (RSSI: -74), connecting
I (4952) BLE2MQTT: Discovered BLE device: c9:92:20:e3:f4:86 (RSSI: -49), not connecting
I (5272) BLE2MQTT: Discovered BLE device: 3f:31:ce:21:1e:6f (RSSI: -55), not connecting
I (5782) BLE2MQTT: Discovered iBeacon broadcaster
I (5782) BLE2MQTT: Discovered BLE device: 80:6f:b0:22:ed:7c (RSSI: -92), not connecting
I (6132) BLE2MQTT: Discovered BLE device: 2b:6f:86:54:56:df (RSSI: -59), not connecting
I (6132) BLE2MQTT: Discovered BeeWi Smart Door broadcaster
E (6142) BT_LOG: Invalid p_eir data.

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4200eaa8  RA      : 0x4200eaa8  SP      : 0x3fcca8f0  GP      : 0x3fc96600
TP      : 0x3fccaac0  T0      : 0x40380e34  T1      : 0x420074ae  T2      : 0xffffffff
S0/FP   : 0x3fcce424  S1      : 0x42008902  A0      : 0x00000000  A1      : 0x00000000
A2      : 0x00000000  A3      : 0x00000005  A4      : 0x00000000  A5      : 0x00000000
A6      : 0xa0000000  A7      : 0x0000000a  S2      : 0x3fcce224  S3      : 0xffffffc5
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00000080  T5      : 0x3fcca77c  T6      : 0x00000001
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000003
MHARTID : 0x00000000

Stack memory:
3fcca8f0: 0x3fc98680 0x3fc9cd58 0x3fcce424 0x420092a6 0x00000086 0x00000054 0xffffffc5 0xffffffc5
3fcca910: 0x3fc98680 0x3fcce224 0x3fcce424 0x420092cc 0x00000054 0x00000056 0x000000df 0xffffffc5
3fcca930: 0x3c134dac 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0x00000040 0x00000000
3fcca950: 0x00000000 0x00000000 0x00000000 0x0000075b 0x0039352d 0x5454514d 0x4542382d 0x65522f34
3fcca970: 0x72617473 0x67690074 0x00006500 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcca990: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcca9b0: 0x4038acd2 0x4038acc6 0x3fccaa30 0x3fc96600 0x3fccaac0 0xffffffff 0x3fca2000 0xffffffff
3fcca9d0: 0x3fcc9a4c 0x00000000 0x00000001 0x00000001 0x3fca2000 0x3fca2000 0x00000001 0x600c0000
3fcca9f0: 0xa0000000 0x00000014 0x3fccaa8c 0x00000000 0xffffffff 0x3fcc9a70 0x00000000 0x00000000
3fccaa10: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fca2000 0x00000001 0x3fca2000 0x00000001
3fccaa30: 0x3fccaa8c 0x00000000 0x3fcc9a4c 0x4038a6c6 0x00000000 0x00000000 0x00000000 0xffffffff
3fccaa50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fccaa70: 0x00000000 0x00000001 0x3fca2000 0x42009526 0x00000000 0x00000000 0x00000000 0x3fcce424
3fccaa90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fccaab0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000150
3fccaad0: 0x3fcca330 0x00000000 0x3fc9bc48 0x3fc9bc48 0x3fccaad0 0x3fc9bc40 0x00000014 0x3fcc9a78
3fccaaf0: 0x3fcc9a78 0x3fccaad0 0x00000000 0x00000005 0x3fcc9acc 0x32656c62 0x7474716d 0x7361745f
3fccab10: 0x0000006b 0x3fccaac0 0x00000005 0x00000000 0x3fca7d48 0x420048b8 0x00000000 0x3fca6148
3fccab30: 0x3fca61b0 0x3fca6218 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x3fccd5d8
3fccab50: 0x42004026 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fccab70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fccab90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fccabb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fccabd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fccabf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fccac10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000028 0x3c135320 0x00001789 0x3fcc0edc
3fccac30: 0x3fc9e0a4 0x3fccac24 0x3fc9e09c 0x00001770 0x00000000 0x42008262 0x3fccac05 0x00000018
3fccac50: 0x00000003 0x3fca8964 0x00000000 0x00000000 0x3fcb1274 0x3fccac48 0x00000010 0x3fccb358
3fccac70: 0x3fcc0014 0x3fcb7201 0x3fccac64 0x00000020 0x00000000 0x00000000 0x00000000 0x00000000
3fccac90: 0x420ef9e4 0x00000000 0x00000000 0x00000000 0x00000020 0x00000000 0x00000000 0x00000000
3fccacb0: 0x00000000 0x420ef9e4 0x00000001 0x3fcb7718 0x3fca1e24 0x00000228 0x00000000 0x00000000
3fccacd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x12121212 0x00000000 0x00000000

ELF file SHA256: 2948408dc

Rebooting...