shmuelzon/esp32-ble2mqtt #202: configure scan window?

ve6rah, 2024-09-07:
Is there a way to configure the scan window? or frequency?

I have an iBeacon that transmits every 1022.5ms (roughly once every second). When I use the KBeacon Android app it updates every second on the app showing that the iBeacon is working as advertised, and when I press the button that changes the UUID it happens on the next update (within a second).

When I use BLE2MQTT it only updates every 30 seconds, and the button press that changes UUID is only captured on the next scan (up to 30 seconds later, or often missed altogether as the UUID change only lasts 20 seconds)

Is there something configurable to change the scanning window of BLE2MQTT so that it can capture quicker updates?

Thanks.

shmuelzon, 2024-09-08:
Hey, this project scans for BLE advertisements constantly (more or less). You can find the exact configuration at: https://github.com/shmuelzon/esp32-ble2mqtt/blob/7bd29fe04a83468c12dba14e28335e660a00b03f/main/ble.c#L21

Note that, [at some point](https://github.com/shmuelzon/esp32-ble2mqtt/commit/c11e9112eeae401cb9b636a963cf67c08d0584a3), I increased the `scan_interval` value as it help stabilizing the application. Without it, IIRC, it would hang often and/or disconnect from Wi-Fi/BLE as both share the same radio.

ve6rah, 2024-09-08:
If it scans continuously, can you think of any other reason it's only publishing an update on this tracker every 30 seconds or so instead of every second or so as the phone app can do?

shmuelzon, 2024-09-08:
Hard to tell without debugging. Maybe it sees the advertisements but doesn't detect them correctly as an iBeacon? Do you see anything in the logs when the iBeacons transmits? Either before or after the button press?