shmuelzon/esp32-ble2mqtt #205: program description and functionalities

Molino69, 2024-10-11:
I've compiled, uploaded and installed the program according all the instructions, and it seems to be working perfectly.
At least, I think it's working perfectly : via the espressif monitor I see the BLE devices being detected, and MQTT message are being sent and received.
It is unclear to me though, what the functionalities of the program are precisely. Can I send MQTT message to the ESP32 with certain instructions ?
It would be great to have some flowchart, or some description on what the program does, what the functionalities are, etc.
Thanks,
Filip

shmuelzon, 2024-10-11:
Hey, most of the interactions with the BLE devices are going to be via MQTT topics and the actual topic name will vary between devices. I suggest you follow all of the topics your devices publishes, find the ones that are of interest to you. When you want to issue a read command, publish to the same topic with a suffix of (by default) `/get`. If you want to issue a write command, do the same with a suffix of (by default) `/set`.

This should all be in the readme file but you're more than welcome to improve it if you find it lacking.

Thanks

Molino69, 2024-10-11:
Thanks for quick and straight forward reply

closed 2024-10-11 by Molino69