shmuelzon/esp32-ble2mqtt #209: Which ESP32 Boards run out-of the-box?

Matthias-Hess, 2024-11-28:
**Question**
Which ESP32 board will run this project without any changes?

**Additional context**
I have an XIAO ESP32 C3 in my drawer, but I am not sure if it will need tinkering to make it work.

shmuelzon, 2024-11-28:
Hey, as this project doesn't rely on any external components connected to the ESP32, I tend to believe any board will work. There's a C3 build in the latest release, you can try it out: https://github.com/shmuelzon/esp32-ble2mqtt/releases

Good luck!

Matthias-Hess, 2024-11-28:
Yes, it worked! I did the following to run it on my XIAO XIAO ESP32 C3

1. installed the espressif IDE, as instructed
2. git clone
3. entered my sid and password in the data/config.json, as well as moqsquitto.org public mqtt server (for testing only)
4. changed the partition table (partitions.csv) because one partition was too small.
```
# Name,   Type, SubType, Offset,   Size
nvs,      data, nvs,     0x009000, 0x004000
otadata,  data, ota,     0x00d000, 0x002000
phy_init, data, phy,     0x00f000, 0x001000
ota_0,    app,  ota_0,   0x010000, 0x190000
ota_1,    app,  ota_1,   0x1A0000, 0x190000
fs_0,     data, spiffs,  0x340000, 0x040000
fs_1,     data, spiffs,  0x380000, 0x040000
```
5. cd into the esp32-ble2mqtt directory
6. idf.py set-target esp32c3
7. idf.py build
8. idf.py -p COM20 flash monitor

Success!

shmuelzon, 2024-11-29:
I don't understand why you needed to increase the application partition size. I had now issues building for the ESP32-C3 and the latest release has an image for it. Which version of ESP-IDF are you using?

In any case, I'm glad you got it working an hope it does what you need~

Matthias-Hess, 2024-12-01:
I just tried again withethe original partition table from github, on a clean machine where I only installed ESP-IDF v5.3.1. Successfully created esp32c3 image. The eror appears consistently, after the message 'Successfully created esp32c3 image.'.
My guess is that perhaps the image can be created successfully but the issue appears later in the build process, which might explain why you did not see the issue (?).

```
Creating esp32c3 image...
Merged 1 ELF section
Successfully created esp32c3 image.
Generated C:/github/esp32-ble2mqtt/build/ble2mqtt.bin
[1274/1274] cmd.exe /C "cd /D C:\github\esp32-ble2mqtt\bui...ion-table.bin C:/github/esp32-ble2mqtt/build/ble2mqtt.bin"
FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size C:/github/esp32-ble2mqtt/build/esp-idf/esptool_py/CMakeFiles/app_check_size
cmd.exe /C "cd /D C:\github\esp32-ble2mqtt\build\esp-idf\esptool_py && C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.3.1/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/github/esp32-ble2mqtt/build/partition_table/partition-table.bin C:/github/esp32-ble2mqtt/build/ble2mqtt.bin"
Error: All app partitions are too small for binary ble2mqtt.bin size 0x1836a0:
  - Part 'ota_0' 0/16 @ 0x10000 size 0x180000 (overflow 0x36a0)
  - Part 'ota_1' 0/17 @ 0x190000 size 0x180000 (overflow 0x36a0)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\github\esp32-ble2mqtt\build\log\idf_py_stderr_output_4648 and C:\github\esp32-ble2mqtt\build\log\idf_py_stdout_output_4648

c:\github\esp32-ble2mqtt>idf.py --version
ESP-IDF v5.3.1
```

Best regards
Matthias

shmuelzon, 2024-12-03:
This is weird... If I build it locally I have 1% free space in the app partitions:
```
[1233/1233] cd /mnt/c/Users/AssafInbal/Documents/Code/esp32-ble2mqtt/build/esp-idf/...-table.bin /mnt/c/Users/AssafInbal/Documents/Code/esp32-ble2mqtt/build/ble2mqtt.bin
ble2mqtt.bin binary size 0x17da00 bytes. Smallest app partition is 0x180000 bytes. 0x2600 bytes (1%) free.
Warning: The smallest app partition is nearly full (1% free space left)!
```
The "official" release builds done as a [CI run](https://github.com/shmuelzon/esp32-ble2mqtt/actions/runs/11288774422/job/31397239211), has 4% free:
```
[1259/1260] cd /app/shmuelzon/esp32-ble2mqtt/build/esp-idf/esptool_py && /opt/esp/python_env/idf5.2_py3.10_env/bin/python /opt/esp/idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /app/shmuelzon/esp32-ble2mqtt/build/partition_table/partition-table.bin /app/shmuelzon/esp32-ble2mqtt/build/ble2mqtt.bin
ble2mqtt.bin binary size 0x1704b0 bytes. Smallest app partition is 0x180000 bytes. 0xfb50 bytes (4%) free.
Warning: The smallest app partition is nearly full (4% free space left)!
```
And for you it's somehow larger. I'm now wondering if the release builds do actually run.

In any case, I agree that it's too close for comfort and I might need to reduce the configuration partitions in favor of the apps'. My main problem with doing so is that it would also break OTA upgrades for the next release...

BTW - Thanks for the tip! It's much appreciated.