shmuelzon/esp32-ble2mqtt #210: How to enable the Access Point for configuration

Matthias-Hess, 2024-12-07:
**Context**
I successfully flashed my XIAO ESP C3 from the Espressif IDF. Before uploading the firmware, I populated the json config with the credentials of my WLAN Network. Now I want to prepare a second device which I want to give to a friend, so that he can use it in his WLAN. I do not have his WLANs SID or password. 

**Question**
How can I configure the device so that it will enter the "Access Point Mode" on first startup, allowing my friend to connect and enter the SSID and Password?

shmuelzon, 2024-12-07:
It's been a long time since I tasted it but leaving the Wi-Fi configuration at its defaults (our empty values) should trigger the AP for configuration.

Matthias-Hess, 2024-12-07:
yes, that worked. I just used the default config file:
```
{
  "network": {
    "wifi": {
      "ssid": "MY_SSID",
      "password": "MY_PASSWORD"
    }
  },
  "mqtt": {
    "server": {
      "host": "192.168.1.1",
      "port": 1883
    },
    "publish": {
      "retain": true
    }
  }
}
```

which started the access point.

closed 2024-12-07 by Matthias-Hess