shmuelzon/esp32-ble2mqtt #211: Fail-Safe Access Point Mode

Matthias-Hess, 2024-12-08:
**Is your feature request related to a problem? Please describe.**  
Currently, users without access to a PC for flashing the device, or those who are not technically adept, face challenges regaining control over their ESP32 running `esp32-ble2mqtt` firmware. Scenarios where this becomes problematic include:

- **WLAN password change:** When the WLAN network password changes, there is no straightforward way to reconnect the device.
- **Switching WLAN networks:** To move the device to a new WLAN, users must configure the new SSID and password before disconnecting from the current network. Failure to do so results in losing access to the device.
- **MQTT server issues:** If the configured MQTT server is unreachable or rejects the credentials, the device may become unusable.
- **Invalid configuration:** Misconfigurations, whether accidental or due to unforeseen issues, can render the device inaccessible.

**Describe the solution you'd like**  
To address these issues, a fail-safe mechanism should be implemented to regain full control of the device under the following circumstances:

- WLAN credentials are incorrect or outdated.
- The SSID of the WLAN changes.
- The configured MQTT server is unreachable or invalid.
- The device configuration is corrupted or otherwise problematic.

**Proposed Solution: Button-Activated Access Point Mode**  

A button on the device could trigger a fail-safe **Access Point (AP) mode**. Here's how it would work:

1. **User Trigger:** Pressing the button starts AP mode regardless of the device's current configuration or state.
2. **Indicator Feedback:** A flashing LED indicates that AP mode is active.
3. **Timeout Mechanism:** If no connection is made to the access point within 2 minutes, the device reboots and resumes normal operation.

This solution offers users a straightforward, hardware-based way to regain control without relying on flashing tools or technical expertise.

**Implementation Notes**  

- **Optional Feature:** The feature should be entirely optional, activated via a build-time switch (e.g., a `#define` directive in the source code).
- **Hardware Constraints:** Not all devices have available pins, buttons, or LEDs. Therefore, the feature must be configurable (at build time only) to adapt to different hardware setups.
- **Source Code Only:** This feature should not be toggled via the `config.json` file to avoid scenarios where an accidental misconfiguration locks the user out of the recovery mechanism.

**Describe alternatives you've considered**  

1. **Automatic AP Mode on WLAN Failure:**  
   Initially, I considered automatically entering AP mode if the device fails to connect to the configured WLAN. However, this approach is flawed because temporary WLAN outages could inadvertently trigger AP mode, disrupting normal operation.

2. **Factory Reset Mechanism:**  
   Another idea was a button for factory reset, but this is riskier as it wipes all settings, making it less user-friendly in comparison to an AP mode for reconfiguration.

**Additional Context**  

This feature would significantly improve usability for non-technical users or those in environments without ready access to flashing tools. A clearly documented setup process for this mode, alongside a warning about the optional hardware requirements, would help ensure smooth adoption.

Matthias-Hess, 2024-12-26:
I wanted to kindly follow up on this enhancement request. It’s been a few weeks since I submitted it, and I understand you may be busy. I’m curious if this feature is something you might consider adding to the project in the future.

If you need more details or clarification, I’d be happy to provide them. Thanks again for your hard work and for maintaining this project—it’s greatly appreciated!

Looking forward to hearing your thoughts.

Best regards,
Matthias

shmuelzon, 2024-12-26:
Hey,

Sorry, must have missed the notification on the first message.
Generally speaking, I do agree that some fallback mechanism in case of Wi-Fi connection failures would be nice.
Using a hardware button that's built in to the code (config/define/other) would be problematic since then we'll start tying into specific hardware whereas, so far, the images are generic since it relies on the ESP chip's peripherals only and nothing external.

So, while not as optimal, starting the AP on Wi-Fi connection failures would probably be our best bet here but, as you pointed out, there are some shortcomings with that approach. I guess something like (if you couldn't connect for X minutes/attempts) we could start the AP and, if no-one connected to it within Y minutes, reboot. Since the early ESP-IDFs had all sorts of issues connecting to Wi-Fi with BLE running there's currently code in place to reset the device when it fails to connect too many times. Perhaps we can use that to start an AP instead and only if no one connects to it, then reboot.

In any case, it's been a while since I actively worked on this project and I'm currently occupied with others.
I'd be more than happy to guide/review a pull request but I'm not sure I'd have the time to work on this myself.

Good day

Matthias-Hess, 2024-12-27:
another (maybe stupid) idea: if the hardware supports USB OTG, it could perhaps allow to edit the config file if mounted as an USB drive on a computer (I don't know if that is feasible without creating multiple forks of the software, which I would like to avoid).

shmuelzon, 2024-12-28:
I think that USG OTG would be part of the different ESP chip default configuration so there's no need to fork it. Maybe there's already a component that exposes the SPIFFS via USB so that's doable. However, that would still mean one needs to remove the hardware from wherever it's at and connect it to a PC. At which point, while using the file explorer is very simple, it's not a long way from using a serial connection or whatever to upload a new configuration.

Still, a nice to have feature.