shmuelzon/esp32-ble2mqtt #212: Can't flash v0.18.1 ble2mqtt-full.esp32-c3.bin to esp32c3.

goddade, 2025-01-20:
*Please make sure you're using the ESP-IDF version specified in the [README](https://github.com/shmuelzon/esp32-ble2mqtt/blob/master/README.md) file, any other version is not supported.*

**Describe the bug**
Can't flash v0.18.1 ble2mqtt-full.esp32-c3.bin to esp32c3.

**To Reproduce**
Steps to reproduce the behavior:
```
PS F:\Downloads> esptool.exe -p COM8 -b 115200  erase_flash
esptool.py v4.7.0
Serial port COM8
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.3)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 60:55:f9:b1:28:28
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 17.4s
Hard resetting via RTS pin...
PS F:\Downloads> esptool.exe -p COM8 -b 115200  write_flash 0 .\ble2mqtt.esp32-c3.bin
esptool.py v4.7.0
Serial port COM8
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.3)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 60:55:f9:b1:28:28
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Unexpected chip id in image. Expected 5 but value was 0. Is this image for a different chip model?

A fatal error occurred: .\ble2mqtt.esp32-c3.bin is not an ESP32-C3 image. Use --force to flash anyway.
```

**Expected behavior**
 Flash firmware to chip.

**Configuration and logs**
If applicable, add your configuration file (without passwords) and logs (preferably of DEBUG level) to help understand your problem.

**Additional context**
Add any other context about the problem here.

shmuelzon, 2025-01-22:
Hey, this sounds similar to #209...
I just [rebuilt](https://github.com/shmuelzon/esp32-ble2mqtt/actions/runs/12915000811/job/36015830634) an image and you can download it from: https://github.com/shmuelzon/esp32-ble2mqtt/actions/runs/12915000811/artifacts/2470087703.

I don't see any errors or warnings in it's generation so it should work properly.

goddade, 2025-01-22:
Similar errors:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004d1fa
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
......
```

goddade, 2025-01-22:
I'll try compiling from source.